michaelbeaumont/dht-sensor #23: Working with DHT22 with a ESP32 board always returns in an TIMEOUT

yudistrange, 2023-06-08:
### What happened?

I am trying to get readings from a DHT22 sensor using my ESP32 dev board. The result is always a timeout.
The board and the sensor are working. I have verified that by flashing LuaRTOS and getting the sensor to respond back with the temperature / humidity readings.

I looked at the specification of the sensor and then your driver. It mostly makes sense. The timeouts are in the correct range as per the sensor specification. I also tried to implement the specification on my own, but still got the same result. I also forked your repo and increased the timeout, just to ensure that the sensor is not slow. I am also only using the `--release` tag for optimized binary as talked about in [this issue](https://github.com/michaelbeaumont/dht-sensor/issues/4)

So I think I may be missing out on some initial setup - setting the pins to be pull-down / pull-up, some other configuration :shrug: 

Are you aware of any such initialization steps that must be performed for the sensor to work?
Here's my [repo](https://github.com/nilenso/immaterial/tree/dht-sensor) that's using your driver.

knightpp, 2023-06-20:
Reproducible on `v0.2.1` and `main` branch.

klimaschkas, 2023-10-29:
For anyone coming along this issue, the solution is to use the GPIO pin in open drain mode:
```rust
let mut sensor = gpio::PinDriver::input_output_od(pin).unwrap();   // instead of input_output(pin).unwrap()
```

michaelbeaumont, 2023-10-29:
Would love to hear whether that solves it for others. This may depend on exactly how the sensor is wired but I think open drain is in fact required for the DHT. It's mentioned/used in both the README and the example.

Ziothh, 2023-12-08:
I have a similar issue on my Raspberry Pi Pico H.

I'm new to embedded and tried multiple crates which keep responding with timeouts. 
Previously used `dht-pio`, which hasn't tested the DHT11 yet so I discovered this crate.

Didn't find the `rp2040_hal` crate referencing configuring pins for open drain so I guessed that `pins.gpio28.into_push_pull_output()` would to the trick.

# My Code
```rust
#![no_std]
#![no_main]

use rp_pico as bsp;

use bsp::{entry, hal::clocks::ClockSource};
use defmt::*;
use defmt_rtt as _;
use dht_sensor::DhtReading;
use embedded_hal::digital::v2::OutputPin;
use panic_probe as _;

use bsp::hal::{clocks::init_clocks_and_plls, pac, sio::Sio, watchdog::Watchdog};

#[entry]
fn main() -> ! {
    let mut pac = pac::Peripherals::take().unwrap();
    let mut watchdog = Watchdog::new(pac.WATCHDOG);
    let clocks = init_clocks_and_plls(
        rp_pico::XOSC_CRYSTAL_FREQ,
        pac.XOSC,
        pac.CLOCKS,
        pac.PLL_SYS,
        pac.PLL_USB,
        &mut pac.RESETS,
        &mut watchdog,
    )
    .ok()
    .unwrap();

    let sio = Sio::new(pac.SIO);
    let pins = rp_pico::Pins::new(
        pac.IO_BANK0,
        pac.PADS_BANK0,
        sio.gpio_bank0,
        &mut pac.RESETS,
    );

    let core = pac::CorePeripherals::take().unwrap();
    let mut delay = cortex_m::delay::Delay::new(core.SYST, clocks.system_clock.get_freq().to_Hz());

    let mut dht11_pin = pins.gpio28.into_push_pull_output();
    match dht11_pin.set_high() {
        Ok(value) => info!("Set GPIO28 HIGH: {}", value),
        Err(err) => error!("Failed to set GPIO2 HIGH: {}", err),
    };

    loop {
        info!("Reading from DHT11");
        match dht_sensor::dht11::Reading::read(&mut delay, &mut dht11_pin) {
            Ok(data) => {
                debug!(
                    "Temp info:\nHumidity: {}\nTemperature: {}",
                    data.relative_humidity, data.temperature
                );
            }
            Err(error) => {
                match error {
                    dht_sensor::DhtError::Timeout => {
                        error!("Timeout");
                    }
                    dht_sensor::DhtError::PinError(_e) => error!("DHT PinError"),
                    dht_sensor::DhtError::ChecksumMismatch => error!("DHT ChecksumMismatch"),
                }
            }
        };

        delay.delay_ms(1000);
    }
}
```

# My sensor setup
[See Pico pin layout](https://www.raspberrypi.com/documentation/microcontrollers/images/pico-pinout.svg)

Data: Green cable from DHT11 left pin to Pico pin 34 (GP28)
Power: Red cable from DHT11 middle pin to Pico pin 36 (3V out)
Ground: Black cable from DHT11 right pin to Pico pin 38 (GND)

[image]

michaelbeaumont, 2023-12-08:
@Ziothh can you try [gpio::InOutPin](https://docs.rs/rp2040-hal/latest/rp2040_hal/gpio/struct.InOutPin.html) as used in [the rp2040 DHT example](https://github.com/rp-rs/rp-hal/blob/28ec5271f83bcbcf2ecbb18bf4bd819b248f1594/rp2040-hal/examples/dht11.rs#L87-L88)?

Ziothh, 2023-12-08:
@michaelbeaumont That works! Thanks for the quick response!

closed 2023-12-08 by michaelbeaumont

SergioGasquez, 2023-12-14:
For reference, here is a working Wokwi project with ESP32: https://wokwi.com/projects/384087992261474305

PR #25 (2023-12-08): docs(README.md): add note about open drain
Closes #23