michaelbeaumont/dht-sensor #26: When in debug mode, the implementation is (presumably) too slow and returns a Timeout (ESP32C3)

reivilibre, 2024-01-28:
It might be nice to include a note in the readme alerting people to the fact that you should try the library in `--release` mode.

In my case (on an ESP32C3), in debug mode, the library does not work. I think this is because the timings are messed up due to poor code performance, but I'm not sure as I didn't research much further. (I already spent a long time wondering if my open-drain config was wrong on this board for some reason; another HAL for it has a dedicated open drain in-out mode which this HAL does not, so I was wondering if it was a shortcoming of this HAL).

(as a note: I think there is a way to override the compilation level for a single dependency crate, so we can do that in our top-level projects if we still want debug mode for some reason)

michaelbeaumont, 2024-01-29:
Sure, do you want to open a PR?

reivilibre, 2024-01-29:
I have opened one.

Thanks for the crate, by the way!

PR #27 (2024-01-29): Add note to README about compiling with `--release` to ensure timing-sensitive code is correct
Closes #26
closed 2024-01-29 by michaelbeaumont